bmellink/IBusBM #7: IBus_Multimonitor - Doesn't send the 2 simulated sensors to my FS-i6 TX

Dmack103, 2020-04-05:
I recently picked up two ESP32devkitC32D microprocessors.  After some headaches trying to get the sketches to upload (capacitor fixed between ground and EN fixed that), I tried the IBus_Multimonitor code.  I made the change to the "IBusServo.begin(Serial1, 1, 21, 22);" line and rem'd out the original line in the code.  My Sensor signal wire has the appropriate diode on pin 17 pin with the black ring on the pin side.  The main part of that wire attaches to pin 16.  The Servo signal wire is hooked up to pin 21. 

I was able to get the Servo readings, but am not sure what I am looking at for the Cnt, Poll, and Sensor.  Cnt seems to count by 11 +/-, Poll seems to increase by about 50 +/- and Sensor = 0.  

While this was running, I checked the transmitter to see if there were any new sensors in the list.  There was not.  I then did some experimenting.  I have two FS-CVT01 voltage sensors and I hooked them in series between the receiver and the harness going to the ESP32.  Those sensors did show up in the TX sensor list.  The output to the serial monitor changed as well.  Cnt is increasing by 9+/-, Poll is at 0, and the sensor is still increasing about 50 +/-.

Does anyone have any words of advice for what I should try next?  I am really looking forward to send custom telemetry back to the transmitter.  Thanks in advance!

bmellink, 2020-04-05:
Let me explain what Cnt, Poll and Sensor mean:
- all counters start at 0 and any value >0 means there is a level of communication that is picked up from the receiver
- cnt-rec is a counter that indicates how many servo values have been received (from TX to RX). This value should increase all the time and is a way to indicate your receiver is working.
- cnt_poll is a counter that indicates how many times the receiver has scanned the bus for the sensors you define. The multimonitor sketch defines 2 dummy sensors: a dummy IBUSS_RPM and a dummy IBUSS_TEMP at the first 2 sensor addresses. The sketch returns an ok, and then receiver asks the sketch which sensor type these sensors are. Normally the receiver then stops polling. In your case it continues to increment, meaning there is an issue
- cnt_sensor is a counter that then should increase for each time the receiver asks the sensor to provide a new value.

In your case the cnt_sensor stays at zero, which is strange. I suspect there is something different in your receiver setup and that the protocol request step is not executed correctly. Can you let me know which receiver type you have?

One explanation could be in the protocol code. If you look at the IBusBM.cpp source file you will see around line 192 the following code:

```
  case PROTOCOL_COMMAND_TYPE: // 0x90, send sensor type
         // echo sensortype command: 0x06 0x91 0x00 0x02 0x66 0xFF 
       stream->write(0x06);
       stream->write(PROTOCOL_COMMAND_TYPE + adr);
       stream->write(sensorType[adr]);
       stream->write(0x02); // always this value - unkwown <---- THIS ONE
       chksum = 0xFFFF - (0x06 + PROTOCOL_COMMAND_TYPE + adr + sensorType[adr] + 2);
```

This value 2 I could not explain from the protocol description and maybe there is something different with your receiver. If you want to experiment with the value here maybe we can get this to work. Change `stream->write(0x02);` to `stream->write(0x0); `
Please note that the calculation of chksum in the next line also needs to be adjusted (the 2 at the end of that line needs to be changed to the same value). Maybe the value should be a 1 or 3. I simply do not know that that byte means in the protocol.

regards. Bart

Dmack103, 2020-04-07:
Thanks for getting back with me Bart!  I am using FS-iA6B receivers.  I pulled another one out of a plane to make sure it was not just a fluke with this one receiver, but both perform the same with Sensor=0.  I also modified the .CPP file to the different settings for 0x02 and the +2 in the following line of code.  I substituted the 2 with 1, 3, and 0.  Each time I saved the .CPP file, I reloaded the Multimonitor sketch and uploaded to the board.  The output is nearly the same for each of these scenarios, with the sensor=0.

I recently loaded the benbojangles latest firmware (10 channels) to the transmitter.  Could this be causing the issue?  What firmware version were you using on your TX?

Thanks,
Dave
[image]

bmellink, 2020-04-07:
I just realized it is probably a hardware issue: If your TX pin from your Arduino board is not connected correctly towards to the FS-iA6B there is no data coming back from the sketch towards the receiver and that would also explain the data you see. You indicated you use a diode, which works in most cases. You can try the following:
- replace a the diode with a resister of 1.2 k Ohm
- add a resister of about 4.7 k Ohm between the connection point where the RX + FS-iA6B data line + diode come together towards the +3.3 Volt power line

I am also wondering if this can be caused by the power supply voltage used. the ESP32devkitC32D is a device that can work with voltages between 2.3 V to 3.6 V and the voltage on the pins should not exceed the supply voltage used (i.e. 3.6 V). The FS-iA6B can only operate between 4 and 6.5 V meaning you can not combine the two without proper voltage shift. In your current setup the RX pin on the ESP board may get a too high voltage from the FS receiver. Can you show how you did the power supply wiring to ensure all devices get the right voltage?

Dmack103, 2020-04-08:
I removed the diode and put in the resistors as described above.  Everything is now working great!  Thanks so much for walking me thru this Bart!

closed 2020-04-08 by Dmack103

bmellink, 2020-04-08:
That is good. I am still a bit worried about the voltage shift circuit missing. Just for protection against overloading the ESP32 serial input you could add a 470 Ohm resistor in the RX line. It is a bit of a poor mans voltage shift but I would put something like that in place or redisign the circuit altogether using an official voltage shift circuit. If you have an oscilloscope you can experiment with the right resistor values by looking at the signals.

Dmack103, 2020-04-08:
Just so I'm clear, you are talking about putting the 470 ohm resistor onto
the servo signal wire?  That way both signal wires will have some voltage
drop since the sensor signal wire is already covered with the 1.2 k ohm and
the 4.7 k ohm resistors, correct?  I will give that a try.  Unfortunately I
don't own an oscilloscope.  Redesigning the circuit on the signal wire side
of things is outside my knowledge realm right now, other than adding
resistors.  I could probably hook a potentiometer to that signal wire and
increase the resistance until the signal isn't read, and then back it off.
Would that work?

bmellink, 2020-04-08:
No. I meant putting the 470 Ohm within the line to the RX pin of the ESP32 (which I think is pin 16 in your setup). So from the receiver you have 3 resistors: one to +3.3V(4.7k), one to the RX pin (470 Ohm) and one to the TX pin (1.2k). But you can also leave it as it is. The overload on the RX pin is probably handled by an internal protection circuit inside the ESP chip.

Dmack103, 2020-04-09:
I can give that a try.  I actually did not have it wired up as you described.  I had the 1.2 k resistor on the 17 pin TX, the 4.7K on the 16 pin RX, and the power in going to the 5v pin on the ESP 32.  Now I understand your concern for the Voltage in on the 3.3 pin.  I am not sure why the 5v pin doesn't show up in the ESP32 data sheet, but it is there and it works (not hooked up
when powering from the USB)  For now, I will leave the Vin as is and change the resistor on the 16 pin from 4.7k to 470 ohm.