bmellink/IBusBM #9: Single-wire mode looks that not working...

vertexod, 2020-05-16:
Trying to use library with single-wire mode. Wiring with arduino according https://github.com/betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry
But it wont work... Can't find any options to use one UART only... :(

`
+---------+
| FS-iA6B |
|         |
| Ser RX  |---|<---\       +------------+
|         |        |       | FC         |
| Sensor  |---[R]--*-------| Uart TX    |
+---------+                +------------+
`

bmellink, 2020-06-07:
I am not sure I fully understand what you are trying to do and what you have tried until now. Your picture is not clear either. Can you please clarify?

Hans007a, 2020-08-18:
With a flight controller You can use IBUS RX with Telemetry on only one wire. Flysky compatible RX Flit10 have a one wire connection for this. User Vertexod give You a link with circuit diagramm for the connection that works with flight controller.
The question is, is this possible with esp32 uart with your library?
My question: Why we can't use ONE uart for both functions? RX ->steering, TX->telemetry?

bmellink, 2020-08-29:
In theory it should be possible to use one wire. It is rather strange the FS-iA6B has split the telemetry from the servo signals in 2 separate iBus pins (actually separate iBus busses). It would have been more logical to use a single pin for all and have a single iBus-bus. 

I have been analyzing the signals and there are 2 problems:
1. sometimes the signal timing overlaps: the servo signals on the servo pin happen at the same time as the polling of the telemetry sensors on the telemetry pin. This will always be a problem
2. the signals from the FS-iA6B do not behave like a true open collector circuit. In an open collector circuit the "1" (+5V) is realized by a fixed resistor and the pins use a transistor to pull down the value to "0" (zero Volt). It looks like the FS-iA6B drives the output high using a transistor, which means that if another device on the bus wants to take ownership of the bus, the device can not drive the bus low (0).
I have been trying various hardware circuits to get this to work, but I could not do this reliable. So I think there are 2 options: (a) use an Arduino with 2 serial interfaces or (b) use a separate arduino for telemetry and another for the servo signals.

closed 2020-11-20 by vertexod

vertexod, 2021-01-12:
@bmellink is it possible to contact with you via mail/telegram/whatsapp? if yes, please, contact with me:
WhatsApp, Viber, telegram: +48883040715 or maybe via FB Messenger: https://www.facebook.com/vertexod
I'm still trying to finish with single-wire connection, and, actually, it already worked on STM32F411 + pure C + HAL, and even the current library worked with single-wire connection on STM32F411 on stm32duino (with small changes), but currently, it shows on stm32duino only 1 sensor (that's with what I'm fighting now)... If you will have time, contact me, and maybe together we could finish the experiments if you interested in it of course. )