bmellink/IBusBM #15: Generic STM32F103C8 Compile error

Ivzo, 2020-09-13:
When i try to run an example i have this:

.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:40:15: error: variable or field 'onTimer' declared void
   40 | void  onTimer(stimer_t *htim) {
      |               ^~~~~~~~
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:40:15: error: 'stimer_t' was not declared in this scope; did you mean 'timer_t'?
   40 | void  onTimer(stimer_t *htim) {
      |               ^~~~~~~~
      |               timer_t
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:40:25: error: 'htim' was not declared in this scope; did you mean 'tm'?
   40 | void  onTimer(stimer_t *htim) {
      |                         ^~~~
      |                         tm
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp: In member function 'void IBusBM::begin(HardwareSerial&, int8_t, int8_t, int8_t)':
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:122:16: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
  122 |         static stimer_t TimHandle; // Handler for stimer
      |                ^~~~~~~~
      |                timer_t
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:123:8: error: 'TimHandle' was not declared in this scope
  123 |        TimHandle.timer = TIMER; // Set TIMx instance.
      |        ^~~~~~~~~
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:124:58: error: 'getTimerClkFreq' was not declared in this scope; did you mean 'getTimerClkSrc'?
  124 |        TimerHandleInit(&TimHandle, 1000 - 1, ((uint32_t)(getTimerClkFreq(TIMER) / (1000000)) - 1)); // Set TIMx timer to 1ms
      |                                                          ^~~~~~~~~~~~~~~
      |                                                          getTimerClkSrc
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:124:8: error: 'TimerHandleInit' was not declared in this scope
  124 |        TimerHandleInit(&TimHandle, 1000 - 1, ((uint32_t)(getTimerClkFreq(TIMER) / (1000000)) - 1)); // Set TIMx timer to 1ms
      |        ^~~~~~~~~~~~~~~
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:125:36: error: 'onTimer' was not declared in this scope
  125 |        attachIntHandle(&TimHandle, onTimer); // Attach onTimer interupt routine
      |                                    ^~~~~~~
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:125:8: error: 'attachIntHandle' was not declared in this scope
  125 |        attachIntHandle(&TimHandle, onTimer); // Attach onTimer interupt routine
      |        ^~~~~~~~~~~~~~~
*** [.pio\build\genericSTM32F103C8\lib273\IBusBM\IBusBM.cpp.o] Error 1

Can you give some advice how to fix?

Ivzo, 2020-09-17:
Found this: In new STM32_core stimer_t replaced by HardwareTimer

bmellink, 2020-09-17:
Thanks Ivzo for finding this issue,

The STM code was provided by another user as I do not have an STM board myself. I suspect this user has some additional included header or uses another version of the core libraries that allow the file to compile correctly. I need to acquire a board and test this myself. For now: If you have any improvements, please forward the code and I will merge this in the main branch.

thanks

Bart

Mstitelp, 2020-09-17:
also a mountain of errors as in the first message. I do not know what to do. blue pill stm32

bmellink, 2020-09-18:
I think I found the issue. It is related to the fact the STM32 code was provided by a user that was using a version of the STM32 library before 1.6. As Ivzo indicated the timer functions are renamed and HardwareTimer should be used. Also the architecture of the timer code is somewhat changed as well. I am rewriting that part of the code now and should have a version for test later today.

bmellink, 2020-09-18:
I just committed a fix. Can you please test this version on your SMT32 modules? If it works ok, we can close this issue and I will update the lib version to v 1.1.5 to be included in the Arduino lib manager

Mstitelp, 2020-09-18:
c:/users/admin/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\FinishVersion.ino.cpp.o: in function `setup':
FinishVersion.ino.cpp:(.text.setup+0x60): undefined reference to `Serial2'
c:/users/admin/appdata/local/arduino15/packages/stm32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: FinishVersion.ino.cpp:(.text.setup+0x68): undefined reference to `Serial3'
collect2.exe: error: ld returned 1 exit status
exit status 1
Ошибка компиляции для платы Generic STM32F1 series.

Mstitelp, 2020-09-18:
the problem of the first post has disappeared. this is what I threw out. I will understand

bmellink, 2020-09-18:
If you have a problem with 'Serial2', it means you are using a board that does not have a 2nd serial port that is defined as 'Serial2'

Mstitelp, 2020-09-18:
I almost figured it out, I think it's little things. thank you for the project and for your work.

Ivzo, 2020-09-18:
Tried to run an example STM32 from examples. Still have some errors:

.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:85:6: error: no declaration matches 'void IBusBM::begin(HardwareSerial&, TIM_TypeDef*, int8_t, int8_t)'
   85 | void IBusBM::begin(HardwareSerial &serial, TIM_TypeDef * timerid, int8_t rxPin, int8_t txPin) {
      |      ^~~~~~
In file included from .pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.cpp:26:
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.h:43:8: note: candidate is: 'void IBusBM::begin(HardwareSerial&, int8_t, int8_t, int8_t)'
   43 |   void begin(HardwareSerial &serial, int8_t timerid=0, int8_t rxPin=-1, int8_t txPin=-1);
      |        ^~~~~
.pio\libdeps\genericSTM32F103C8\IBusBM\src\IBusBM.h:40:7: note: 'class IBusBM' defined here
   40 | class IBusBM {
      |       ^~~~~~
*** [.pio\build\genericSTM32F103C8\lib0a0\IBusBM\IBusBM.cpp.o] Error 1

Lines that have errors (hihtlited)
Line 85:   void **IBusBM**::**begin**(HardwareSerial &serial, TIM_TypeDef * timerid, int8_t rxPin, int8_t txPin) {
and
Line 109:  if (!IBusBMfirst && timerid **!=** IBUSBM_NOTIMER) {

bmellink, 2020-09-18:
Are you sure you have the latest version of the STM32 core libraries
installed?
If you go to Tools->Board Manager and scroll down to "STM32 Cores", this
should be version 1.9.0 or higher. If you have an older version and you can
not upgrade to 1.9.0 you may have the wrong URL configured below the
"Additional board manager URLs" setting. This should be:
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

hope this helps

regards

Bart Mellink

Mstitelp, 2020-09-19:
Ivzo, as I understood, you can not initialize ?? HERE

IBusServo.begin(Serial,0,PA3,PA2);
IBusSensor.begin(Serial1,0,PB11,PB10);

Ivzo, 2020-09-19:
Sorry, it was my mistake. That's why i deleted the message. Now everything compiles. Thank you.

closed 2020-09-20 by bmellink