bmellink/IBusBM #16: IBusBM not compatible with megaavr

haruss77, 2020-10-28:
Hello Bart
Do you plan to make this library compatible with megaavr boards such as nano every?

bmellink, 2020-10-28:
I just made a small change to support MegaAVR. I do not have a board, so I can't test it, but it should compile now. Can you please test it? Please note the timer function is not supported, so you should use IBUSBM_NOTIMER and call the iBus.loop() yourself.

haruss77, 2020-10-28:
Hello Bart
Wow that was fast... 
Unfortunately there are still ERRORS while compiling... (see below). I have to say that I am not that skilled in programming... Sorry. I hope the ERROR messages make sense for you..

BR

Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano Every, None (ATMEGA4809)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\bless\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\bless\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\bless\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=off -vid-pid=2341_0058 -ide-version=10813 -build-path C:\Users\bless\AppData\Local\Temp\arduino_build_780594 -warnings=none -build-cache C:\Users\bless\AppData\Local\Temp\arduino_cache_779495 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\bless\Documents\Arduino\FlySky\TraktorTest\TraktorTest.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\bless\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\bless\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\bless\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=off -vid-pid=2341_0058 -ide-version=10813 -build-path C:\Users\bless\AppData\Local\Temp\arduino_build_780594 -warnings=none -build-cache C:\Users\bless\AppData\Local\Temp\arduino_cache_779495 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\bless\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\bless\Documents\Arduino\FlySky\TraktorTest\TraktorTest.ino

Using board 'nona4809' from platform in folder: C:\Users\bless\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6

Using core 'arduino' from platform in folder: C:\Users\bless\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6

Detecting libraries used...

"C:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\nona4809" "C:\\Users\\bless\\AppData\\Local\\Temp\\arduino_build_780594\\sketch\\TraktorTest.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for IBusBM.h: [IBusBM@1.1.4]

ResolveLibrary(IBusBM.h)

  -> candidates: [IBusBM@1.1.4]

"C:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\bless\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\ [... cut ...]
C:\Users\bless\Documents\Arduino\libraries\IBusBM\src/IBusBM.h:43:8: note: candidate: void IBusBM::begin(int&, int8_t, int8_t, int8_t)

   void begin(HardwareSerial &serial, int8_t timerid=0, int8_t rxPin=-1, int8_t txPin=-1);

        ^~~~~

C:\Users\bless\Documents\Arduino\libraries\IBusBM\src/IBusBM.h:43:8: note:   no known conversion for argument 1 from 'UartClass' to 'int&'

Using library IBusBM at version 1.1.4 in folder: C:\Users\bless\Documents\Arduino\libraries\IBusBM 

exit status 1

no matching function for call to 'IBusBM::begin(UartClass&)'

bmellink, 2020-10-28:
The error `error: reference to 'HardwareSerial' is ambiguous` means you are still using the original IBusBM.h

Are you sure you replaced all changed files with the versions I changed? The files that I changed are: IBusBM.h, IBusBM.cpp and library.properties. They are located on your computer in:
`C:\Users\bless\Documents\Arduino\libraries\IBusBM\`
 and 
`C:\Users\bless\Documents\Arduino\libraries\IBusBM\src`
Once you replaced the files with the ones from github you need to restart the Arduino environment (exit/start Arduino) so Arduino checks the library property files. This should get rid of the error: 
`WARNING: library IBusBM claims to run on avr, esp32, stm32, mbed architecture(s) and may be incompatible with your current board which runs on megaavr architecture(s).`

haruss77, 2020-10-28:
Hello Bart

Of course I did not update them in the mentioned directory ;-). In the Arduino-SW I went just to Tools/Manage Libraries and updated the IBusBM.h. That was not enough...

The program can now be compiled, having only one red entry left in the bottom status window.
avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description

I changed one of your examples "Ibus_multimonitor" so that it works with my old nano. It works fine with the old one. Fine means that I get values from different channels of the RC-receiver. But the very same program wont work for the nano every. Not a single value is received from the RC-receiver... (program see zip-file attached)
[Ibus_multimonitor_Nano-adjust.zip](https://github.com/bmellink/IBusBM/files/5454985/Ibus_multimonitor_Nano-adjust.zip)

bmellink, 2020-10-28:
As indicated in my first answer: The timer function is not supported, so you should use IBUSBM_NOTIMER and call the iBus.loop() yourself from within loop(). See the readme for how that is done.

haruss77, 2020-10-29:
Hello Bart
I tried it without success so far. I dont get any values from the nano every. Same code with my old nano works perfect. Due to my limited programming-knowledge it is most likely a programming / syntax error. 
It would be great if you could have a look at it (see ZIP-file attached)
BR
[Ibus_multimonitor_Nano-adjust (2).zip](https://github.com/bmellink/IBusBM/files/5457274/Ibus_multimonitor_Nano-adjust.2.zip)

bmellink, 2020-10-29:
Your loop() does not call the IBusServo.loop() function every millisecond. You can try the following code to make this work:

```
void loop() {
  int cnt=0;
  while (1) {
    IBusServo.loop(); // ensure we call the loop every millisecond
    delay(1);
    if (cnt++ > 500) {
      // do something else once every 500 ms
      cnt=0;    
      int val=0;
      // show first 8 servo channels
      for (int i=0; i<8 ; i++) {
        // call internal loop function to update the communication to the receiver 
        val = IBusServo.readChannel(i);
        Serial.print(val);
        Serial.print(" ");
      }
      Serial.print("Cnt=");
      Serial.print(IBusServo.cnt_rec); // count of how many times servo values have been updated
      Serial.print(" POLL=");
      Serial.print(IBusSensor.cnt_poll); // count of polling for sensor existance
      Serial.print(" Sensor=");
      Serial.println(IBusSensor.cnt_sensor); // count of polling for sensor value
    }
  }
}
```

haruss77, 2020-10-29:
Hello Bart

Still the same. Your code works fine with the old nano but not with the nanao every. The wireing is the same at both nanos. RX0 is used for the IBUS connection.

haruss77, 2020-10-29:
Forgot to attach the code...
[Ibus_multimonitor_Nano-adjust_Bart.zip](https://github.com/bmellink/IBusBM/files/5457504/Ibus_multimonitor_Nano-adjust_Bart.zip)

bmellink, 2020-10-29:
Just see another thing: I think you are using the wrong serial port in setup():

```
// iBUS setup
  IBusServo.begin(Serial, IBUSBM_NOTIMER);

```
The EVERY uses Serial for talking to the USB and Serial1 for talking to your RX/TX pins. Can you try changing the line in setup() to
```
// iBUS setup
  IBusServo.begin(Serial1, IBUSBM_NOTIMER);
```

bmellink, 2020-10-29:
If this works, you can actually also enable Serial2 for sending sensor data back to your controller. The Arduino Nano EVERY is based on the AVR 4809 processor. This processor has 4 hardware UART ports. 

The standard Arduino libraries only have 2 ports enabled: serial through the USB port and serial1 through 
pins TX/RX pins. This article https://forum.arduino.cc/index.php?topic=639740.0 describes a way to change 
the core library routines to also enable serial2 and serial3. This allows you to provide both 
servo output and sensor input using a single small sized Arduino module. Please note if you start doing this, you should repeat this change each time you let Arduino update the core libraries.

haruss77, 2020-10-29:
Whohooo! Heureka! :-)

What a big difference a little number can make.

Thank you Bart for your strong support...

closed 2020-10-29 by haruss77