bmellink/IBusBM #17: Support for Arduino Due

itavero, 2020-11-02:
Unfortunately the library currently doesn't support the Arduino Due.
{code}
WARNING: library IBusBM claims to run on avr, esp32, stm32, mbed architecture(s) and may be incompatible with your current board which runs on sam architecture(s).
/Users/arnom/Documents/Arduino/libraries/IBusBM/src/IBusBM.cpp: In member function 'void IBusBM::begin(HardwareSerial&, int8_t, int8_t, int8_t)':
/Users/arnom/Documents/Arduino/libraries/IBusBM/src/IBusBM.cpp:91:36: error: no matching function for call to 'HardwareSerial::begin(int, UARTClass::UARTModes)'
     serial.begin(115200, SERIAL_8N1);
                                    ^
/Users/arnom/Documents/Arduino/libraries/IBusBM/src/IBusBM.cpp:91:36: note: candidate is:
In file included from /Users/arnom/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/Arduino.h:195:0,
                 from /Users/arnom/Documents/Arduino/libraries/IBusBM/src/IBusBM.cpp:25:
/Users/arnom/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/HardwareSerial.h:29:18: note: virtual void HardwareSerial::begin(long unsigned int)
     virtual void begin(unsigned long);
                  ^
/Users/arnom/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino/HardwareSerial.h:29:18: note:   candidate expects 1 argument, 2 provided
exit status 1
Error compiling for board Arduino Due (Native USB Port).
{code}

Seems like there's something wrong with the Arduino abstraction for the Due. 🤔

PS: I'm aware that later on I will probably need a level translator to deal with the 3V3 vs 5V signals.

bmellink, 2020-11-05:
I did not have the time yet to look into supporting the Due and I do not have a device to test. If I make changes to the code, are you able to run tests?

itavero, 2020-11-05:
I can perform some tests. If I find some time myself, I'll see if I can fix it and open up a PR.

itavero, 2021-03-29:
Finally had some time to have a look.

I noticed that the serial port configuration (`SERIAL_8N1`) passed to `Serial.begin` is actually the default option (according to the [Arduino website](https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/)).
Just removing the parameter should fix it, I think.

I'll see if I have some time in the next week or so to verify this.

bmellink, 2021-03-31:
You are right this should be an easy fix if this is the only change needed. Would love to hear if that is all that is required.

itavero, 2021-04-04:
Just removing the argument at least makes it compile/run again.

Unfortunately it seems that I might be running into a hardware issue with the shield I've designed.
Used an TXB0108 IC to do the level translation between 5V (from the receiver) and the `IOREF` of the Arduino board.
Unfortunately I do see the UART signals with a scope on the 5V side, but I don't see them on the `IOREF` (3V3 in case of the Due) side.

In other words, I've not (yet) been able to verify that the data communication is actually working.

MotoMotoRo, 2022-01-02:
Happy Birthday !!!
I tried this library on STM32F103 and no example works..It could have been something wrong ...!

MotoMotoRo, 2022-01-17:
Hello.
I use the blue pill STM32F103 and FlySky FS-iA6 + IA6b.
I tried the IBusBM-1.1.4 library, the examples with servo work, but those with sensor, telemetry can't solve. Can help be made?
Thanks.

id3vi5er, 2022-01-17:
> Hello.
> I use the blue pill STM32F103 and FlySky FS-iA6 + IA6b.
> I tried the IBusBM-1.1.4 library, the examples with servo work, but those with sensor, telemetry can't solve. Can help be made?

>What do you mean by can't solve? It doesn't compile? Could you show a log?

MotoMotoRo, 2022-01-17:
Thanks for the reply.
Sorry for the inaccuracy.
Compiles, but no sensor appears on the FS-i6 display.
Maybe I need to add or change something and I don't know.
Thank you again.