bmellink/IBusBM #18: ibus_sensor does not work on NANO

Dmitrykrz, 2020-11-09:
This example works fine with arduino mega.
I want to run it on nano. So I change 
` // initialize serial port for debug`
 ` Serial.begin(115200);`
`  // iBUS connected to serial1`
`  IBus.begin(Serial1);`
to 
` // initialize serial port for debug`
`  //Serial.begin(115200);`
`  // iBUS connected to serial`
 ` IBus.begin(Serial);`
Usb cable is not connected.
In this case it does not work. I see no sensors on the transmitter. 
Is nano supported for sensor only operation?

id3vi5er, 2020-11-09:
It works with a nano, i use it myself  you just need to Programm ist via
the 6 Pin header and a ISP, in my case a Myavrlight (: then the tx/rx works
just fine, but not in combination with I2C or other hardware timing related
things, I suppose...

id3vi5er, 2020-11-09:
And you cannot run the Serial and the iBus on the same port, doesn't work either ;(

Dmitrykrz, 2020-11-09:
> Just get the Serial.begin() out of the code and it should work....
This is exactly what I did, look at my code.
Why I need to program it via ISP? To get rid of arduino bootloader?

closed 2021-03-31 by bmellink