bmellink/IBusBM #20: Error with PWM

jan5650, 2020-12-08:
Dear Bmellink!

I have a problem with this library. 
It works great until i want to read PWM values with PulseIn()

[image]

As you see in the picture i used 2 input pin connected to a Reciever channel1 and channel2.
When i read transmission's centered axis PWM values with  pulseIn(), then i got something like this:

1460
1458
1466
1420
1102         <---- big bounce!!!
1466
1459

When i comment out IBus.begin(Serial3) the problem vanished! Then i got a smooth PWM values aroung 1460 without big bounces.

I thin think this bounce happens when IBUS polling sensors. Or not?

What i am doing wrong?
The rest of the program there is IBus.setSensorMeasurement. The IBUS values are OK! Just the PWM bouncing.
Timer issue? Interrupt issue?

Thank you!

bmellink, 2020-12-08:
Hi Jan

This absolutely has to do with the interrupts. Let me explain: the IBusBM library attaches code to a timer interrupt to handle the sbus communication. This gives max freedom to write whatever you want in the loop() function. The disadvantage is that you can not guarantee code in the loop() function always runs at exactly the same speed. The pulsiIn function become highly unreliable when interrupts are used. Please note that the Serial.print() function also uses an interrupt based communication and may also influence the pulseIn() function.

There are 2 different approaches to fix this in your case:

Approach (1): disable the use of interrupts by IbusBM by using the IBUSBM_NOTIMER) option and write your loop() function in such a way that you call ibus.loop() at least once per millisecond.  You can then use a counter to call your pulseIn() function only once per 5 seconds or so. This is how it will look:

```
void setup() {
  .... other setup code goes here ....
  IBus.begin(Serial3, IBUSBM_NOTIMER);  
}

void loop() {
  static int cnt=0;
  // .... other loop() code that needs to be called at a high frequency goes here ....
  if (cnt++>5000) {
     // here goes code that only needs to be called once per 5 seconds
     input1 = pulseIn(2, HIGH);
     // do something with input1
     cnt = 0;
 }
  // code to call the polling function. Should normally be called at least once every 1 ms
  Ibus.loop();  
  wait(1);
}

```

Approach (2):  Rewrite the pulseIn() function as an interrupt driven function. You would need to attach an interrupt to the PIN you use to calculate the pulse width (not all PINS on your Arduino board support interrupts). A quick Google search describes a similar approach here: https://forum.arduino.cc/index.php?topic=367449.0

jan5650, 2020-12-10:
Thank you for the quick answer!

I will try tomorrow as your first suggested. My arduino IDE won't recognize
wait() function. Is there any library that includes it? Or is it a simple
Delay()? In the loop I read GPS sentences, I don't want to miss any, so the
delay is not so good for me :(
And I want to call the pulsein more often like 5sec... Is this problem? And
what happens when I call the IBUS pulls often like once per sec?

Thank you for the answer!

Bart Mellink <notifications@github.com> ezt írta (időpont: 2020. dec. 8.,
K, 23:54):

Mstitelp, 2020-12-13:
Hello. I have a similar story with stm32. When I try to save data to flash, after erasing the page, the servos start jumping. IBUS_NOTIMER does not completely cure the problem, partially. Breaking my head

jan5650, 2020-12-13:
Ibus.loop(); works for you?

I do as Bmellink suggest, so i use IBus.begin(Serial3, IBUSBM_NOTIMER);  in the Setup section.
In this case the PulseIn() function works well, there is no PWM signal jump, its meaured good.

BUT, when i call Ibus.loop(); happens nothing... There is no incoming sensor data. I used in loop, used in void, neither work... 
I call this function good?

Mstitelp, 2020-12-13:
Ibus.loop (); I started to tune and I got some data on the screen, but the servos kept jumping and I abandoned this function. can you somehow change the timer ?? I'm a student in programming

jan5650, 2020-12-13:
**Maybe Bmellink soon be here, and save our lives :D**

bmellink, 2020-12-13:
Hi Jan and Mstitelp,

Can you please post your code? It is a bit hard to see what you are doing without the code. Also let me know which Arduino chip you are running this on. 

Please note there is a description of the NOTIMER option in the README file. Please take a look there too to ensure you are doing the right thing.
In case IBUS_NOTIMER does not fully cure the problem you need to look at other libraries you use that also use interrupts. 

Overall, if your sketch needs to do more than one thing at a time and also needs to be time critical, you should design it using interrupts for the time critical things. Pulsein(0 was designed to be used on installations that do nothing else.  This means you should rewrite the pulsein() routine as interrupt driven. (which I described in my first answer as option 2).

The same is true for some of the servo libraries out there: Some of these libraries do not handle interrupts very well and other do. The result is that servos start to jitter.

jan5650, 2020-12-14:
**Hi Bmelink!**

This is my code (i removed the irrelevant lines), i used Arduino Mega 2560. I used GPS on UART Serial1, and IBUS on UART Serial3, there is two more input pin: 2 and 9. I want to read PWM values at this two input pins. I used Flysky FS-IA10B reciever. I get all the GPS inputs, gets all PWM values good, without jumpings, but there is no incoming sensor data :(

--------------------------------------------------------------------------------------------
`#include <NMEAGPS.h>
#include <GPSport.h>
#include <IBusBM.h>

static NMEAGPS  gps;

//int ****;
//...
//            Declarate a lot variables
//...
//int ****;

IBusBM IBus;

**static void doSomeWork( const gps_fix & fix )**
     **{**
        value1 = pulseIn(2, HIGH);
        value2 = pulseIn(9, HIGH);

        IBus.loop();

         //Doing things with this variables
        //Doing things with this variables
       //Doing things with this variables
**}**

**void setup()**
**{**
      pinMode(10, OUTPUT);
      pinMode(11, OUTPUT);
      pinMode(2, INPUT);
      pinMode(9, INPUT);

      IBus.addSensor(IBUSS_DIST, 1);
      IBus.addSensor(IBUSS_MODE, 2);
      IBus.addSensor(IBUSS_ETET, 3);
      IBus.addSensor(IBUSS_SPE, 4);
      IBus.begin(Serial3, IBUSBM_NOTIMER);
      gpsPort.begin(9600);
**}**

**void loop()**
**{**
     gps.available( gpsPort );
     doSomeWork( gps.read() );
**}`**

--------------------------------------------------------------------------------------------------------------------

**When i remove IBUSBM_NOTIMER from IBus, then there is incoming Sensors data, all works great, but the measured PWM signals become jumpy :( Am i used good the loop function in the posted code? What do i wrong?**

Thank you for the answer!

bmellink, 2020-12-14:
I suggest 3 changes:
1. Put the IBus.begin(Serial3, IBUSBM_NOTIMER); statement before the IBus.addSensor(IBUSS_DIST, 1); command. since IBus.begin() initiates variables that could be used in addSensor(). In your case it should still work, but I can not guarantee the code continues to work in a next version of the library.

2. Remove the IBus.loop(); from the doSomeWork(); the location is incorrect as doSomeWork() will only be called when data has been received from the gps receiver

3. your loop() function is incorrect: you should call IBus.loop() more often. This is better:

```
void loop() {
  static unsigned int cnt=0;
  if (cnt++>2000 &&  gps.available( gpsPort )) {
     // the gps coordinates will be retrieved once per 2 seconds
    doSomeWork( gps.read() );
     cnt = 0;
 }
  Ibus.loop();  
  delay(1);
}
```

Please test if this works. You can optimize this some more by splitting the doSomeWork() function in a function specific for the gps coordinates and one for each of the pulseIn() commands:

```
void doSomeGpsWork( arg ) {
  //. handle the gps data and send back using setSensorMeasurement()
}

void DoSomePin2Work() {
  value1 = pulseIn(2, HIGH);
  // handle the value1 data and send back using  using setSensorMeasurement()
}

void DoSomePin9Work() {
  value2 = pulseIn(9, HIGH);
  // handle the value2 data and send back using  using setSensorMeasurement()
}

void loop() {
  // we call each of the 3 sensors once per 3 seconds: the first second the gps, the second the pulse on pin2 and then pin9
  static unsigned int cnt=0;
  switch(cnt) {
    case 0:
        while (gps.available( gpsPort ))   doSomeGpsWork( gps.read() );
        break;
    case 1000:
        DoSomePin2Work();
        break;
    case 2000:
        DoSomePin9Work();
        break;
  }
  Ibus.loop();  
  if (cnt++>3000) cnt=0;
  delay(1);
}
```

jan5650, 2020-12-15:
Thank you for the suggestions!

Sadly i cannot split the doSomeWork() function as you wrote, because i need
fast realtime datas. In this case i get the correct GPS coordinates just
after every 2-3 sec, and if i pull the switch on transmitter at cnt=0 then
i need to wait about 2 sec to get readed wich is too slow.

Previously i used the Ibus.loop(); function in the main loop, but still no
incoming sensor data :(
Delay(1) is necessary?

Bart Mellink <notifications@github.com> ezt írta (időpont: 2020. dec. 14.,
H, 22:22):

Mstitelp, 2020-12-15:
Hi jan! I called the ibus loop function after each line of code and on my screen they were initialized with zeros, there was no data. Then I dropped this feature. I'll post my code later, I want BMellink to look. I have a large code, it needs to be optimized to make it easier to watch it.

jan5650, 2020-12-15:
I have no data too :(

Mstitelp <notifications@github.com> ezt írta (időpont: 2020. dec. 15., K,
8:12):

bmellink, 2020-12-15:
Jan

Can you please explain what you are trying to do and why you need the real time data? Most GPS receivers using NMEA only provide a new data point once per second. Maybe there is an easier way to do it. And why do you need to read the pulse values on PIN 2 and 9 in the first place? What pulse is it? Is this a switch on your tranceiver? There are easier ways to do that.
I have some other code in mind how you could rewrite the pulseIn() function
regards

Bart

jan5650, 2020-12-15:
Thank you for the reply!

I used you library to store and show sensore data from Distance between the
Vehicle actual coordinate and from the base GPS coordinate . I want to
print Distance once per a sec on Flysky FS-i6 transmitter, so i need to
poll the sensors AT LEAST once per sec. And i want to change target
position, save positon, etc. with 2 piece Flysky Transmitter's Switch. I
want to react to switch fast, so i need to read PWM pulse's at least once
per sec too. It is doable?

Thank you! I very like you Supportive behavior!

Bart Mellink <notifications@github.com> ezt írta (időpont: 2020. dec. 15.,
K, 15:32):

bmellink, 2020-12-15:
So you are using PulseIn() to read the PWM pulse from the receiver? Why can't you use the IbusBM library also to get the switch data from your transciever? You need one more serial port on the 2560 but you have 4 ports anyway: 1 for GPS, 2 for IbusBM and still 1 left for debug. see the Ibus_multimonitor example (https://github.com/bmellink/IBusBM/blob/master/examples/Ibus_multimonitor/Ibus_multimonitor.ino)

bmellink, 2020-12-15:
If you only need to establish if a switch is ON or OFF on a PWM signal, you can actually also do this in hardware by adding 2 resistors and a capacitor to the output of the PWM signal and using an analog read to read the voltage of the pin

Mstitelp, 2020-12-15:
Hi BMellink !!
Look at this please. STM32F103C8T6, a radio controlled vehicle that drives on GPS points. Made it possible to save points to flash in order to return to them again. Sometimes, when saving a point to flash, the servos begin to behave inadequately, then when I save the point again, everything is restored and works fine. this is the frequency. when I write ibus_notimer and want to save coordinates, the servos sometimes jerk for a short time (I'm fine with that), but there is no telemetry

#include <Ultrasonic.h>
#include <Servo.h>
#include <IBusBM.h>
#include <TinyGPS++.h>
Ultrasonic ultrasonic (PB3,PB4);
IBusBM IBusServo;
IBusBM IBusSensor;
TinyGPSPlus GPS;
Servo ServoSteeringWheel;
Servo ServoMotor;
Servo Bunker1; 
HardwareSerial SerialServo(PA3,PA2);
HardwareSerial SerialSensor(PB11,PB10);
HardwareSerial Gps(PA10,PA9);
const uint32_t LatArrey[] = {0x801D400,0x801D800,0x801DC00,0x801E000,
                             0x801E400,0x801E800,0x801EC00,0x801F000,
                             0x801F400,0x801F800,0x801FC00};
const uint32_t LongArrey[] = {0x801D404,0x801D804,0x801DC04,0x801E004,
                              0x801E404,0x801E804,0x801EC04,0x801F004,
                              0x801F404,0x801F804,0x801FC04};
int valBunker1;
int valPoint; 
int valShine; 
int valServoMotor;
int valServoSteeringWheel; 
int RoundUp1;
int RoundUp2; 
int i;
float bearing1;  
float bearing2; 
float distance1; 
float distance2;  
float distance3; 
float llat1;  
float llong1;  
float llong2; 
float llat2;
float llong3; 
float llat3;  
float angleGrad; 
bool savePoint; 

void setup() {

pinMode(PA8,OUTPUT);
SerialServo.begin(115200);
SerialSensor.begin(115200);
Gps.begin(9600);
IBusServo.begin(SerialServo);
IBusSensor.begin(SerialSensor);
ServoSteeringWheel.attach(PB7,1000,2000); 
ServoMotor.attach(PB6,1000,2000); 
Bunker1.attach(PB9,1000,2000);
IBusSensor. addSensor(IBUS_MEAS_TYPE_CLIMB_RATE);
IBusSensor. addSensor(IBUS_SENSOR_TYPE_ACC_Y);
IBusSensor. addSensor(IBUS_SENSOR_TYPE_ACC_Z);
IBusSensor. addSensor(IBUS_SENSOR_TYPE_GPS_STATUS);
IBusSensor. addSensor(IBUS_SENSOR_TYPE_GROUND_SPEED);
}

void loop() {
  
  valPoint = IBusServo.readChannel(7); // get latest value for servo channel 8
  if(valPoint < 1000){
    valPoint = 1000;
  }
  valPoint = (valPoint - 1000) / 100;
  llat1 = (float(*(__IO uint32_t *)(LatArrey[valPoint]))) / 10000000; 
  llong1 = (float(*(__IO uint32_t *)(LongArrey[valPoint]))) / 10000000; 
  llat2 = (float(*(__IO uint32_t *)(LatArrey[0]))) / 10000000; 
  llong2 = (float(*(__IO uint32_t *)(LongArrey[0]))) / 10000000; 
  while (Gps.available() > 0)
  GPS.encode(Gps.read());
  llat3 = GPS.location.lat();
  llong3 = GPS.location.lng();
  distance1 = ......................................
  distance2 = .....................................
  distance3 = mathematical calculation
  angleGrad = .....................................
  IBusSensor.setSensorMeasurement(1,valPoint + (int(angleGrad / 6) * 100));
  IBusSensor.setSensorMeasurement(2,distance2 * 100);
  IBusSensor.setSensorMeasurement(3,ultrasonic.read()); 
  IBusSensor.setSensorMeasurement(4,GPS.satellites.value());
  IBusSensor.setSensorMeasurement(5,GPS.speed.kmph() * 1000);

  valServoSteeringWheel = IBusServo.readChannel(0); // get latest value for servo channel 1
  ServoSteeringWheel.writeMicroseconds(valServoSteeringWheel); 
  valServoMotor = IBusServo.readChannel(1); // get latest value for servo channel 2 
  ServoMotor.writeMicroseconds(valServoMotor); 
  valShine = map(IBusServo.readChannel(2),1000,2000,0,255);  // get latest value for servo channel 3
  analogWrite(PA8,valShine);
 
  valBunker1 = IBusServo.readChannel(5); // get latest value for servo channel 6
  Bunker1.writeMicroseconds(valBunker1);
  
  if(IBusServo.readChannel(9) < 1500){
    savePoint = true;    
  }
  
  if(IBusServo.readChannel(9) > 1900){  // get latest value for servo channel 10
    while(savePoint){   
     if(Gps.available() > 0){
     GPS.encode(Gps.read());
     //if(GPS.location.isUpdated()){
       
        HAL_FLASH_Unlock();  
        
        while (FLASH->SR & FLASH_SR_BSY);
        if (FLASH->SR & FLASH_SR_EOP) {
            FLASH->SR = FLASH_SR_EOP;
        }

        FLASH->CR |= FLASH_CR_PER;
        FLASH->AR = LatArrey[valPoint];
        FLASH->CR |= FLASH_CR_STRT;
        while (!(FLASH->SR & FLASH_SR_EOP));
        FLASH->SR = FLASH_SR_EOP;
        FLASH->CR &= ~FLASH_CR_PER;

       
      HAL_FLASH_Program(0x02U, LatArrey[valPoint], uint32_t((GPS.location.lat() * 10000000)));
      
      HAL_FLASH_Program(0x02U, LongArrey[valPoint], uint32_t((GPS.location.lng() * 10000000)));
      
      HAL_FLASH_Lock ();
      
      savePoint = false;
     }
   }
  }
}

bmellink, 2020-12-16:
hi Mstitelp ,

I hope I can give you some pointers. The erratic behavior of servos may not be due to IbusBM but has to do with the stm32.

Three things:
(1) If you want to use IBUS_NOTIMER you have to call the correct loop() function. This depends on the order for which you call the begin() functions for your 2 instances of IbusBM. In your case you call:

```
void setup() {
    // ....
    IBusServo.begin(SerialServo);
    IBusSensor.begin(SerialSensor);
    ///...
}

```
which means you have to call the IbusSensor.loop() function in your main routine. Internally, this one will then call the IbusServo.loop() function.

(2) I am not very familiair with the stm32 and how it uses the Flash, but I do believe it will block all interrupts if you are writing to the same memory block of the flash where the current program is executed from. This is not what you want, because both IbusBM and the servo routines depend on interrupts. This would explain the erratic behavior of the servos when you write to the flash. I am not sure why you need to write to the same memory sector of your code (I believe the statement HAL_FLASH_Program(0x02U, LatArrey[valPoint], uint32_t((GPS.location.lat() * 10000000))); does this).

I found this application note from stm on this topic: https://www.st.com/resource/en/application_note/dm00311483-eeprom-emulation-techniques-and-software-for-stm32-microcontrollers-stmicroelectronics.pdf. See paragraph 4.7. Maybe one way to solve it is to relocate your program to RAM memory on load. There are multiple articles on addressing this. Here is one: https://www.embeddedrelated.com/showthread/comp.arch.embedded/225536-1.php and https://electronics.stackexchange.com/questions/327858/why-execute-code-from-ram

You can also consider using another board or having your EEPROM external for storing your GPS coordinates. See https://www.hobbytronics.co.uk/arduino-external-eeprom

3. One other work around could be to turn off your servos during the flash update. it is a kind of poor mans solution though.

Mstitelp, 2020-12-24:
Hi jan5650. I won the problem. I can tell you if you are still suffering

jan5650, 2020-12-24:
> 
> 
> Hi jan5650. I won the problem. I can tell you if you are still suffering

Sure! I am full Ears!

Mstitelp, 2020-12-24:
I threw out the servo library and wrote a delay () control, then I wrote the magic word IBUS_NOTIMER and called a loop after each line. the fact is that these libraries work on interruption and interfere with each other, and the code also starts to work badly. I still had one problem, based on your code, I think you will not have it. if you have any questions - write

jan5650, 2020-12-28:
Thank you!

The IBUS_NOTIMER will be good, but sadly i cannot call it. By the way i can
call it, but there is no incoming data

Mstitelp <notifications@github.com> ezt írta (időpont: 2020. dec. 24., Cs,
22:35):

jan5650, 2021-01-13:
Dear Bmellnik!

I did it! I was stupid... Your library support IBUS Servo's. I didn't know that it can read all channel's PWM signal trought 1 IBUS Servo cable... Your library works great! I removed the cables from all channel output, and put only one cable to the IBUS Servo and i can read all of the channels WITHOUT error's. No jumping, no error, just the exact PWM value! GREAT! thank you! 👍 

By the way are you familiar in the Flysky FS-i6 transmitter firmware?
Yesterday i saw a Flysky FS-I6 remote, which LCD diplay's background light is lit much more that the original FS-i6, and when i power on the remote there is no security check for the switches/axis. I think it can specified in the firmware.
I got your lates source code, i tried to search in the /Sources/Sources files, but don't know witch file i need to modify

Thank you for all!

closed 2021-03-31 by bmellink