bmellink/IBusBM #21: Arduino uno - telemetry - cnt_sensor stays 0

jverhuls, 2021-02-02:
I have an arduino uno and a FS-ia6b. I'm trying to send telemetry data towards my transmitter (and only telemetry data).
I'm using the code as is and the sketch Ibus_sensor.ino where i remove all the Serial.print and Serial.begin commands.

I have a wire like described (tested it with my multimeter if the resistor is working on the TX pin. I used TX on port 1 and RX on port 0 but I see no telemetry data. 

If i'm using altsoftserial (on port 8,9) for debugging i see the cnt_poll increasing, but the cnt_sensor stays zero.
I tried with a 1,2K resistor and 4,7K resistor but non work.
Any clue on what I can try or what the problem could be?

bmellink, 2021-03-31:
Hi jverhuls,

Sorry for the late reply. I did not notice this was still an open issue. I looked into your problem and I think it is due to the Arduino Uno board. If you look at the schematics of the UNO board at https://content.arduino.cc/assets/UNO-TH_Rev3e_sch.pdf you see 2 resistors RN4A and RN4B that feed the Serial signal (TX/TX) to the onboard USB device. These resistors "fight" with the combined open collector signal towards the FS-ia6b.

The 1k resistor RN4B is keeping the serial signal high (too close to +5V) while the Ibus library is trying to send information back by pulling your 1,2k or 4,7k resistor low. A value of 4,7k will never work. In case you are using an 1.2k resistor the digital high value will be +5V and the digital low value will be about +2.7V, which is not low enough for the FS-ia6b to recognise as a digital 0.

There are 4 things you can try:
1. Do not use a resistor, but use a diode (as described in the original readme). This will ensure the digital 0 is below the +0.7V which should be low enough
2. Use another Arduino board that does not have an embedded USB port on the same serial line you are trying to use
3. Reduce the value of the external resistor to 330 Ohm. This will reduce the digital 0 to +1,24V which may just be good enough for the FS-ia6b. I am not sure this will work though.
4. (not recommended, unless you know what you are doing) If you have one of the older UNO boards where the ATMega328 still fits in a IC socket, you can remove the chip, bend the serial ports on pins 2 & 3 outwards and plug back the chip into the socket. You will then need to solder the FS-ia6b to the ATMega328 pins 2(=RX) & 3(=TX). This should then work good enough with an external resistor of 1,2k. However, in order to re-program the UNO you need to temporary disconnect the FS-ia6b and connect the ATMega328 pins to the digital pins 0 and 1 using 2 wires (ATMega328 pin 2 will go to digital pin 0 and ATMega328 pin 3 will go to digital pin 1).

Let me know if this works.

Frehd, 2023-08-11:
Hi,

I found your reply very helpful.

I also tried to use an arduino UNO and a FS-ia6b and faced the same issues when using a resistor.
However switching to a diode solved these for me.

I hope this helps.