bmellink/IBusBM #25: Teensy Support

nicholaskillin, 2022-05-17:
Hey! I'm working on a project that uses IBusBM to receive data from my RC receiver and translate that into commands for DFPlayer. This allows me to play sounds for my RC R2-D2. I got this working pretty good with a Nano.

However, the DFPlayer also needs UART, so I am attempting to move over to a TeensyLC board, which has more than enough UART ports for the RC receiver, DFPlayer, and computer connection.

I have some sample code going, but right now I'm not getting anything but `0` when calling `readChannel` on `IBus`. After some looking around, I'm thinking it likely has to do with the fact that the [Teensy LC uses an unsupported architecture](https://www.pjrc.com/store/teensylc.html). 

I'd be happy to help you get this supported if possible. I don't know anything about Arduino libraries other than loading and using them, but I am a programmer and have an TeensyLC that I'd be more than happy to test with. 

Any chance this is something you'd be willing to support?

bmellink, 2022-05-17:
Hi Nick,

I did a quick check of the Teensy LC board and it should not be a large
effort to support this board. I suspect the problem is calling the right
serial port routines and linking the right timer. Teensy indeed seems to
have enough ports and timers.

A library is just a bunch of C code, so you can look at the source files to
see how they work. The IBusBM lib is really not that complicated.
The main file is this one:
https://github.com/bmellink/IBusBM/blob/master/src/IBusBM.cpp
as you see there are a number of "IFDEF" statements that switch on or off
code depending on the selected architecture
Maybe you want to give it a shot and if it does not work I may have time in
the future to buy a Teensy and try myself.

The quicker way to communicate is by using my direct email: ***@***.***

regards

Bart Mellink

nicholaskillin, 2022-05-19:
Most of my experience is in Ruby and JS, so I'm not sure how helpful I'll be in C, but I have learned a lot of the basics writing my sketch.

If I get some time I'll take a look, but most of what you said is stuff that I'll have to learn in order to make any meaningful contribution.