bmellink/IBusBM #29: Adding RPi Pico support

avielbr, 2022-10-12:
Hi there, thanks for this great library. I'd love to contribute and add support for the RPi Pico RP2040. I'm somewhat new to this -- would you mind pointing out the main things that need to happen in order to get things working with this board, and I'll have a crack at it?

Jakob0854, 2023-04-15:
Hi, I would be also interested in your Pico Integration. Maybe you can dm me on my discord, linked in my Github profile

avielbr, 2023-04-16:
> Hi, I would be also interested in your Pico Integration. Maybe you can dm me on my discord, linked in my Github profile

Hey! Sure, let's talk. I am having trouble finding you on Discord for some reason. Would you mind sending me a DM at avielbr#2096? Thanks.