bmellink/IBusBM #36: nano not work for ibus

hohonorbika, 2023-05-03:
hi i have a problem with get it  work the ibusBM with the nano , i have upload the ino to the uno board as well and its work as should but when i try to do with the nano  its not at all , i did wired up as should, checked many times even used diferent nano ,, brand new,, let say about 4 of them , and still no result ,same issues , any help woud be great thanks

Cobalt6700, 2023-06-15:
Can you confirm your setup? The code would also be helpful. 

I have had this working on MEGAs, UNOs, NANOs and NANO EVERYs with no issues.

hohonorbika, 2023-06-19:
Basically I used the example sketch from the page , I have connected the Ibus ,,servo,, signal pin to rx on the nano , led on nano was popped up when transmitter on , I believe its mean receve something.  I have mentioned ibusbm.begin(Serial); on the void setup , i did ad as well with out and with out serial.begin too but nothing happened,  I tried to print out any data but nothing coming up on the  serial monitor , but if I upload the sketch to the uno board it's work as should with out any problem . I'm a beginner for coding,  just let u know lol so defo I did /do something silly,  but any help would be much appreciated.  Regards

Cobalt6700, 2023-06-19:
Please post your code (using the add code tag - looks like "<>") so we can have a look.

hohonorbika, 2023-06-19:
Working at moment but  as on as I back home I will post it , thank u

hohonorbika, 2023-06-19:
"<#include <Arduino.h>
#include "IBusBM.h"
#include <Servo.h>

IBusBM IBus;    // IBus object
Servo hopper1;  // create servo object to control a servo
Servo hopper2;  // create servo object to control a servo
Servo spiral;
Servo camera;

int navigation_led = 3;       
int hopperled1 = 11;      
int hopperled2 = 12;

void setup() {
  IBus.begin(Serial);    // iBUS csatlakoztatva a  RX pinhez
 
  pinMode(10,OUTPUT);
  pinMode(12,OUTPUT);
  pinMode(3,OUTPUT);
 spiral.attach(6); 
  camera.attach(7); 
 hopper1.attach(8);     // 1es ajto
 hopper2.attach(9);     // 2es ajto
}

void loop() {
  

  int val,val1,val2;
 
  
val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
  analogWrite(3, val);
  
  val = IBus.readChannel(2);  // 3as channel
  val2 = IBus.readChannel(4); // 5os channel
  val1 = IBus.readChannel(5); // 6os channel
  val = IBus.readChannel(6);  //7es channel
  
 hopper1.writeMicroseconds(val1);   // sets the servo position 
  hopper2.writeMicroseconds(val2);   // sets the servo position 
  spiral.writeMicroseconds(val);
  camera.writeMicroseconds(val);
  
if ( val1 >1500){
  digitalWrite(11,HIGH);
  delay(100);
    digitalWrite(11,LOW);
    delay(100);
}
  if ( val2 >1500){
  digitalWrite(12,HIGH);
  delay(100);
    digitalWrite(12,LOW);
    delay(100);
  }
}>"

hohonorbika, 2023-06-19:
`

"<#include <Arduino.h>
#include "IBusBM.h"
#include <Servo.h>

IBusBM IBus;    // IBus object
Servo hopper1;  // create servo object to control a servo
Servo hopper2;  // create servo object to control a servo
Servo spiral;
Servo camera;

int navigation_led = 3;       
int hopperled1 = 11;      
int hopperled2 = 12;

void setup() {
  IBus.begin(Serial);    // iBUS csatlakoztatva a  RX pinhez
 
  pinMode(10,OUTPUT);
  pinMode(12,OUTPUT);
  pinMode(3,OUTPUT);
 spiral.attach(6); 
  camera.attach(7); 
 hopper1.attach(8);     // 1es ajto
 hopper2.attach(9);     // 2es ajto
}

void loop() {
  

  int val,val1,val2;
 
  
val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
  analogWrite(3, val);
  
  val = IBus.readChannel(2);  // 3as channel
  val2 = IBus.readChannel(4); // 5os channel
  val1 = IBus.readChannel(5); // 6os channel
  val = IBus.readChannel(6);  //7es channel
  
 hopper1.writeMicroseconds(val1);   // sets the servo position 
  hopper2.writeMicroseconds(val2);   // sets the servo position 
  spiral.writeMicroseconds(val);
  camera.writeMicroseconds(val);
  
if ( val1 >1500){
  digitalWrite(11,HIGH);
  delay(100);
    digitalWrite(11,LOW);
    delay(100);
}
  if ( val2 >1500){
  digitalWrite(12,HIGH);
  delay(100);
    digitalWrite(12,LOW);
    delay(100);
  }
}>"`

hohonorbika, 2023-06-19:
`#include <Arduino.h>
#include "IBusBM.h"
#include <Servo.h>

IBusBM IBus; // IBus object
Servo hopper1; // create servo object to control a servo
Servo hopper2; // create servo object to control a servo
Servo spiral;
Servo camera;

int navigation_led = 3;
int hopperled1 = 11;
int hopperled2 = 12;

void setup() {
IBus.begin(Serial); // iBUS csatlakoztatva a RX pinhez

pinMode(10,OUTPUT);
pinMode(12,OUTPUT);
pinMode(3,OUTPUT);
spiral.attach(6);
camera.attach(7);
hopper1.attach(8); // 1es ajto
hopper2.attach(9); // 2es ajto
}

void loop() {

int val,val1,val2;

val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
analogWrite(3, val);

val = IBus.readChannel(2); // 3as channel
val2 = IBus.readChannel(4); // 5os channel
val1 = IBus.readChannel(5); // 6os channel
val = IBus.readChannel(6); //7es channel

hopper1.writeMicroseconds(val1); // sets the servo position
hopper2.writeMicroseconds(val2); // sets the servo position
spiral.writeMicroseconds(val);
camera.writeMicroseconds(val);

if ( val1 >1500){
digitalWrite(11,HIGH);
delay(100);
digitalWrite(11,LOW);
delay(100);
}
if ( val2 >1500){
digitalWrite(12,HIGH);
delay(100);
digitalWrite(12,LOW);
delay(100);
}
}`

hohonorbika, 2023-06-19:
<#include <Arduino.h>
#include "IBusBM.h"
#include <Servo.h>

IBusBM IBus; // IBus object
Servo hopper1; // create servo object to control a servo
Servo hopper2; // create servo object to control a servo
Servo spiral;
Servo camera;

int navigation_led = 3;
int hopperled1 = 11;
int hopperled2 = 12;

void setup() {
IBus.begin(Serial); // iBUS csatlakoztatva a RX pinhez

pinMode(10,OUTPUT);
pinMode(12,OUTPUT);
pinMode(3,OUTPUT);
spiral.attach(6);
camera.attach(7);
hopper1.attach(8); // 1es ajto
hopper2.attach(9); // 2es ajto
}

void loop() {

int val,val1,val2;

val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
analogWrite(3, val);

val = IBus.readChannel(2); // 3as channel
val2 = IBus.readChannel(4); // 5os channel
val1 = IBus.readChannel(5); // 6os channel
val = IBus.readChannel(6); //7es channel

hopper1.writeMicroseconds(val1); // sets the servo position
hopper2.writeMicroseconds(val2); // sets the servo position
spiral.writeMicroseconds(val);
camera.writeMicroseconds(val);

if ( val1 >1500){
digitalWrite(11,HIGH);
delay(100);
digitalWrite(11,LOW);
delay(100);
}
if ( val2 >1500){
digitalWrite(12,HIGH);
delay(100);
digitalWrite(12,LOW);
delay(100);
}
}/>

hohonorbika, 2023-06-20:
`<pre>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">Arduino</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;IBusBM.h&#34;</font>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">Servo</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#000000">IBusBM</font> <font color="#000000">IBus</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; IBus object</font>
<b><font color="#d35400">Servo</font></b> <font color="#000000">hopper1</font><font color="#000000">;</font> &nbsp;<font color="#434f54">&#47;&#47; create servo object to control a servo</font>
<b><font color="#d35400">Servo</font></b> <font color="#000000">hopper2</font><font color="#000000">;</font> &nbsp;<font color="#434f54">&#47;&#47; create servo object to control a servo</font>
<b><font color="#d35400">Servo</font></b> <font color="#000000">spiral</font><font color="#000000">;</font>
<b><font color="#d35400">Servo</font></b> <font color="#000000">camera</font><font color="#000000">;</font>

<font color="#00979c">int</font> <font color="#000000">navigation_led</font> <font color="#434f54">=</font> <font color="#000000">3</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;
<font color="#00979c">int</font> <font color="#000000">hopperled1</font> <font color="#434f54">=</font> <font color="#000000">11</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;
<font color="#00979c">int</font> <font color="#000000">hopperled2</font> <font color="#434f54">=</font> <font color="#000000">12</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">IBus</font><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><b><font color="#d35400">Serial</font></b><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; iBUS csatlakoztatva a &nbsp;RX pinhez</font>
 
 &nbsp;<font color="#d35400">pinMode</font><font color="#000000">(</font><font color="#000000">10</font><font color="#434f54">,</font><font color="#00979c">OUTPUT</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#d35400">pinMode</font><font color="#000000">(</font><font color="#000000">12</font><font color="#434f54">,</font><font color="#00979c">OUTPUT</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#d35400">pinMode</font><font color="#000000">(</font><font color="#000000">3</font><font color="#434f54">,</font><font color="#00979c">OUTPUT</font><font color="#000000">)</font><font color="#000000">;</font>
 <font color="#000000">spiral</font><font color="#434f54">.</font><font color="#d35400">attach</font><font color="#000000">(</font><font color="#000000">6</font><font color="#000000">)</font><font color="#000000">;</font> 
 &nbsp;<font color="#000000">camera</font><font color="#434f54">.</font><font color="#d35400">attach</font><font color="#000000">(</font><font color="#000000">7</font><font color="#000000">)</font><font color="#000000">;</font> 
 <font color="#000000">hopper1</font><font color="#434f54">.</font><font color="#d35400">attach</font><font color="#000000">(</font><font color="#000000">8</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; 1es ajto</font>
 <font color="#000000">hopper2</font><font color="#434f54">.</font><font color="#d35400">attach</font><font color="#000000">(</font><font color="#000000">9</font><font color="#000000">)</font><font color="#000000">;</font> &nbsp;&nbsp;&nbsp;&nbsp;<font color="#434f54">&#47;&#47; 2es ajto</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;

 &nbsp;<font color="#00979c">int</font> <font color="#d35400">val</font><font color="#434f54">,</font><font color="#000000">val1</font><font color="#434f54">,</font><font color="#000000">val2</font><font color="#000000">;</font>
 
 &nbsp;
<font color="#d35400">val</font> <font color="#434f54">=</font> <font color="#d35400">map</font><font color="#000000">(</font><font color="#000000">IBus</font><font color="#434f54">.</font><font color="#000000">readChannel</font><font color="#000000">(</font><font color="#000000">8</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">1000</font><font color="#434f54">,</font> <font color="#000000">2000</font><font color="#434f54">,</font> <font color="#000000">0</font><font color="#434f54">,</font> <font color="#000000">255</font><font color="#000000">)</font><font color="#000000">;</font> <font color="#434f54">&#47;&#47;9es channel led szabalyzo</font>
 &nbsp;<font color="#d35400">analogWrite</font><font color="#000000">(</font><font color="#000000">3</font><font color="#434f54">,</font> <font color="#d35400">val</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
[... cut ...]
 &nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">100</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#d35400">digitalWrite</font><font color="#000000">(</font><font color="#000000">12</font><font color="#434f54">,</font><font color="#00979c">LOW</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#d35400">delay</font><font color="#000000">(</font><font color="#000000">100</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">}</font>
<font color="#000000">}</font>

</pre>`

hohonorbika, 2023-06-20:
`[code]

#include <Arduino.h>
#include "IBusBM.h"
#include <Servo.h>

IBusBM IBus;    // IBus object
Servo hopper1;  // create servo object to control a servo
Servo hopper2;  // create servo object to control a servo
Servo spiral;
Servo camera;

int navigation_led = 3;       
int hopperled1 = 11;      
int hopperled2 = 12;

void setup() {
  IBus.begin(Serial);    // iBUS csatlakoztatva a  RX pinhez
 
  pinMode(10,OUTPUT);
  pinMode(12,OUTPUT);
  pinMode(3,OUTPUT);
 spiral.attach(6); 
  camera.attach(7); 
 hopper1.attach(8);     // 1es ajto
 hopper2.attach(9);     // 2es ajto
}

void loop() {
  

  int val,val1,val2;
 
  
val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
  analogWrite(3, val);
  
  val = IBus.readChannel(2);  // 3as channel
  val2 = IBus.readChannel(4); // 5os channel
  val1 = IBus.readChannel(5); // 6os channel
  val = IBus.readChannel(6);  //7es channel
  
 hopper1.writeMicroseconds(val1);   // sets the servo position 
  hopper2.writeMicroseconds(val2);   // sets the servo position 
  spiral.writeMicroseconds(val);
  camera.writeMicroseconds(val);
  
if ( val1 >1500){
  digitalWrite(11,HIGH);
  delay(100);
    digitalWrite(11,LOW);
    delay(100);
}
  if ( val2 >1500){
  digitalWrite(12,HIGH);
  delay(100);
    digitalWrite(12,LOW);
    delay(100);
  }
}
[/code]`

Cobalt6700, 2023-06-20:
@hohonorbika Can you confirm you have this exact code running and working on an UNO? 

There are some problems with your code that I would suggest you sort first before troubleshooting ibus issues. 

Pin 10 is not being used: 

`pinMode(10,OUTPUT);`

Rather than calling the pin number, it may help to call the variable that you have assigned to that pin number. This will also save you work in the future should your pin number change.

`pinMode( hopperled1, OUTPUT );`

You are only initialising 3 variables for getting the ibus channel readings:

`  int val,val1,val2; ` 

however you are reading and writing 5 channels:

```
  val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
  analogWrite(3, val);
  val = IBus.readChannel(2);  // 3as channel
  val2 = IBus.readChannel(4); // 5os channel
  val1 = IBus.readChannel(5); // 6os channel
  val = IBus.readChannel(6);  //7es channel
```

To prevent unwanted behaviour, I would suggest that you create a variable for each channel. 

You are also using delay to blink LED's. I am not sure how IbusBM runs with delay - I would guess not that well since it is using timers in the Library.
I would suggest trying without this code included and see the outcome. 

I would also recommend you have a look at 'blink without delay' in the Arduino examples. Typically, using a delay should be avoided unless specifically required to block other actions.

hohonorbika, 2023-06-20:
Yes I can confirm it's run on uno with out any problem or blocking even with delay

hohonorbika, 2023-06-20:
Yes just realised now about the 10

Cobalt6700, 2023-06-20:
Okay, thanks for confirming. 

Can you confirm that the NANO's you are using have an ATmega328P chip? 
I am not sure of a reason why code working on a UNO would not work on a NANO as they share the same chip and pinout. 

By the looks of things, only using the Servo part of IbusBM shouldn't have a problem with delays.

You might like to include this in your code, it may help with troubleshooting: 

```
//...
const byte num_ibus_chan = 14; // Put the number of ibus channels you are using here. 
conat byte chan_hysteresis = 2;

// rc channel input array
  volatile int in_rc_ch[num_ibus_chan] = { };

// rc channel input array previous
  int in_rc_ch_prev[num_ibus_chan] = { };
//...

void loop(){

Serial.print(F("Cnt=")); Serial.print(IBus.cnt_rec); // count of how many times servo values have been updated

for (byte i = 0; i < num_ibus_chan; i++) {
     in_rc_ch[i] = IBus.readChannel(i);
}

for (byte i = 0; i < num_ibus_chan; i++) {
    if ( in_rc_ch[i] > ( in_rc_ch_prev[i] + chan_hysteresis ) || in_rc_ch[i] < ( in_rc_ch_prev[i] - chan_hysteresis )  ){
        Serial.print(F("RC Ch: ")); Serial.print( i + 1 ); Serial.print(F(" uS: ")); Serial.println( in_rc_ch[i] );
    }          
}

//Your other code here
//...

delay(500);
}      

```

hohonorbika, 2023-06-20:
Yes , I will do , and let u know how was thx

hohonorbika, 2023-06-20:
I did tried nothing happened

Cobalt6700, 2023-06-20:
I ran your sketch, with some modifications, code here:

```
#include <Arduino.h>

#include "IBusBM.h"

#include <Servo.h>

IBusBM IBus;    // IBus object

Servo hopper1;  // create servo object to control a servo

Servo hopper2;  // create servo object to control a servo

Servo spiral;

Servo camera;

const byte num_ibus_chan = 14; // Put the number of ibus channels you are using here. 
const byte chan_hysteresis = 2;

// rc channel input array
  volatile int in_rc_ch[num_ibus_chan] = { };

// rc channel input array previous
  int in_rc_ch_prev[num_ibus_chan] = { };

int navigation_led = 3; 

int hopperled1 = 11;  

int hopperled2 = 12;

int hopperservo1 = 8;

int hopperservo2 = 9;

int spiralservo = 6;

int cameraservo = 7;

void setup() {

pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH);

IBus.begin(Serial);    // iBUS csatlakoztatva a  RX pinhez

Serial.println(F("Sketch Starting"));

pinMode(hopperled1,OUTPUT);

pinMode(hopperled2,OUTPUT);

pinMode(navigation_led,OUTPUT);

spiral.attach(spiralservo);

camera.attach(cameraservo);

hopper1.attach(hopperservo1);     // 1es ajto

hopper2.attach(hopperservo2);     // 2es ajto

Serial.println(F("Sketch Started"));

}

void loop() {

digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));

Serial.print(F("Cnt=")); Serial.println(IBus.cnt_rec); // count of how many times servo values have been updated

for (byte i = 0; i < num_ibus_chan; i++) {
     in_rc_ch[i] = IBus.readChannel(i);
}

for (byte i = 0; i < num_ibus_chan; i++) {
    if ( in_rc_ch[i] > ( in_rc_ch_prev[i] + chan_hysteresis ) || in_rc_ch[i] < ( in_rc_ch_prev[i] - chan_hysteresis )  ){
        Serial.print(F("RC Ch: ")); Serial.print( i + 1 ); Serial.print(F(" uS: ")); Serial.println( in_rc_ch[i] );
    }          
}

for (byte i = 0; i < 14; i++) {
    in_rc_ch_prev[i] = in_rc_ch[i];
}
 

int val,val1,val2, val3, val4;

 

val = map(IBus.readChannel(8), 1000, 2000, 0, 255); //9es channel led szabalyzo
analogWrite(3, val);
val3 = IBus.readChannel(2);  // 3as channel
val2 = IBus.readChannel(4); // 5os channel
val1 = IBus.readChannel(5); // 6os channel
val4 = IBus.readChannel(6);  //7es channel

 

hopper1.writeMicroseconds(val1);   // sets the servo position 

 hopper2.writeMicroseconds(val2);   // sets the servo position 

 spiral.writeMicroseconds(val3);

 camera.writeMicroseconds(val4);

 

if ( val1 >1500){

 digitalWrite(hopperled1,HIGH);

 delay(100);

   digitalWrite(hopperled1,LOW);

   delay(100);

}

 if ( val2 >1500){

 digitalWrite(hopperled2,HIGH);

 delay(100);

   digitalWrite(hopperled2,LOW);

   delay(100);

 }

delay(500);

}

```

This is the output, wiggled some sticks:

```
Sketch Starting
Sketch Started
Cnt=0
Cnt=65
RC Ch: 1 uS: 1500
RC Ch: 2 uS: 1501
RC Ch: 3 uS: 1472
RC Ch: 4 uS: 1500
RC Ch: 5 uS: 1525
RC Ch: 6 uS: 1737
RC Ch: 7 uS: 1500
RC Ch: 8 uS: 1500
RC Ch: 9 uS: 1500
RC Ch: 10 uS: 1500
RC Ch: 11 uS: 1500
RC Ch: 12 uS: 1500
RC Ch: 13 uS: 1500
RC Ch: 14 uS: 1500
Cnt=184
Cnt=45
RC Ch: 3 uS: 1238
Cnt=6
RC Ch: 3 uS: 1412
Cnt=123
RC Ch: 3 uS: 1847
Cnt=240
RC Ch: 3 uS: 1978
Cnt=101
RC Ch: 3 uS: 1462
Cnt=218
RC Ch: 2 uS: 1882
RC Ch: 3 uS: 1466
Cnt=79
RC Ch: 2 uS: 1176
Cnt=196
RC Ch: 2 uS: 1969
Cnt=57
RC Ch: 2 uS: 1501
```

hohonorbika, 2023-06-20:
nothing change , looks like i have ordered a realy bad nano clone maybe , i will order a new one ,,hope fuly original or working one ,, and try with that  thank u fo9r your time and help , as on as is arived the new one will let u know thanks again

hohonorbika, 2024-01-16:
All sorted