bmellink/IBusBM #42: Arduino R4 Minima not supported

boc811, 2023-10-27:
Hi, unfortunately I can't get my code running with Arduino R4 Minima Board. 
I get the following errors message:

../src/IBusBM.h:43:14: error: reference to 'HardwareSerial' is ambiguous
   void begin(HardwareSerial &serial, int8_t timerid=0, int8_t rxPin=-1, int8_t txPin=-1);

megacct, 2023-10-31:
Hi, you could try changing line 32 of IBusBM.h to add your board:

#if defined(ARDUINO_ARCH_MBED) || defined(ARDUINO_UNOR4_MINIMA)

n.b. timer won't work so you'll need to set IBUSBM_NOTIMER and call the loop yourself

boc811, 2023-11-01:
Hello Megactt, thanks so much. Adding  || defined(ARDUINO_UNOR4_MINIM to the ibusBM.h worked out so far. The compilation took quite a while, but at the end the compiler just reported some warnings and no more errors. Now I am going to exchange  the arduino Uno with the Arduino Uno R4 mimima in my hardware project. I'm quite excited how the outcome will be. The exchange of the boards will probably take a week. I'll keep you updated and I am going to post a video on youtube of my project called "Circle Heli" just for fun and share it with you. Cheers, Björn

boc811, 2023-11-03:
Hello Megactt, 
unfortunately your proposal didn't work out with the connected HW (FS-IA6b Receiver ) at the end.
The enclosed test code works fine with an Arduino Uno, but with an Arduino R4 minima it shows only  0's for Channel 0 instead of 1500. However thanks for your support so far. 
Regards, Björn

[Ibus_test.txt](https://github.com/bmellink/IBusBM/files/13255216/Ibus_test.txt)

megacct, 2023-11-04:
Timer should be disabled in the begin call:  IBus.begin(Serial, IBUSBM_NOTIMER); 

Not sure if this will make a difference as you are calling the loop yourself already

PaulNTU, 2024-04-27:
I have just created a branch that should work with your Arduino R4 and put a pull request in for the main library if that helps. This uses the FspTimer to run the loop at 1000hz

You can find the repository with the code here 
https://github.com/PaulNTU/IBusBM/tree/Renesas-Support