bmellink/IBusBM #43: SofwareSerial

lucashudson2002, 2023-12-09:
Is it possible to use SoftwareSerial instead of HardwareSerial? I am using the reciever FS-IA6B with an Arduino Nano, wich has only one Serial.
I tried just changing in the sources but it didn't work.

svdrummer, 2023-12-09:
Why would you use serial? What format are you using? If Ibus, then use the Ibus library'
FS-1A6B doesn't have serial out. Data port: PWM / PPM / i.bus / s.bus

lucashudson2002, 2023-12-09:
The servo and sens pins use Serial to comunicate. So much that in the example multimonitor the IBusSensor and IBusServo begin you pass a Serial as a parameter.
But if im wrong, how can I use IBusSensor anda IBusServo whith an arduino nano?

chupocro, 2024-03-31:
Here is the [video](https://youtu.be/HRapwUg_jeo) of someone reading IBUS with `AltSoftSerial` and here is the [code](https://github.com/playfultechnology/arduino-animatronic/tree/master/AnimatronicController_iBus).

MotoMotoRo, 2024-04-06:
lucashudson2002...For Nano....use https://github.com/pedrocastelani/SoftIBus