bmellink/IBusBM #45: IBUSBM_NOTIMER ... problem.

MotoMotoRo, 2024-04-27:
Hello.
I tried on ESP32 Devkit v1 IBusServo and IBusSensor, they work ok together...until I enter...IBUSBM_NOTIMER.
At that moment the telemetry disappears.
Something can be done!
This is the code.
``#include <IBusBM.h>
// ================================== IBus Servo =============================
IBusBM IBusServo;
HardwareSerial Serial_1(1);
// ================================ IBus Telemetry =============================
IBusBM IBusSensor;

void setup()
{
  Serial.begin(115200);

  IBusServo.begin(Serial_1, IBUSBM_NOTIMER);
  //IBusServo.begin(Serial_1);
  Serial_1.begin(115200, SERIAL_8N1, 4, 15);
  delay(500);
  IBusSensor.begin(Serial2);
  delay(500);

  IBusSensor.addSensor(0x0b);
}
void loop()
{
  IBusServo.loop();                                                                    // Call the Ibus loop to check for Ibus messages from the receiver
  Serial.println(IBusServo.readChannel(0));
  IBusSensor.setSensorMeasurement(1, 33);
}

fanfanlatulipe26, 2024-04-27:
Your loop write a **lot** of data on the serial monitor at **full speed** .... and kills the processor ...

Maybe you should try to only write on serial and/or update the sensor when there is a change in the data (see the ibus2PWM example)

When you use a timer,  the IBusBM::loop(void) in the library is called in an ISR context,  but when you don't use the timer the loop is called by you in a "standard" context and it may explain the behavior of the test.

MotoMotoRo, 2024-04-28:
Hello.
Don't think that ...Serial.println(IBusServo.readChannel(0));
there is a lot of data... No ... IBUSBM_NOTIMER, telemetry Ok, any data I would write on the serial. The problem is if something can be done when it is used, so that it no longer causes the problem.
Sorry for the Google translation.

MotoMotoRo, 2024-05-06:
Solved...I no longer need to use... IBUSBM_NOTIMER.

MotoMotoRo, 2024-05-25:
Hello.
I solved.
Works ok, on AVR, ESP32, STM32 Blue Pill.
Correct and what fanfanlatulipe26 said.
The only problem for it to work on STM32 is that it depends on the IDE version and how it is set.

closed 2024-05-25 by MotoMotoRo