bmellink/IBusBM #49: ESP32 Timer Issue

yousif20121, 2024-07-29:
The code can't be uploaded on ESP32 (NodeMCU & DOIT KIT V1).

`c:\Users\Kio Store\Documents\Arduino\libraries\IBusBM\src\IBusBM.cpp: In member function 'void IBusBM::begin(HardwareSerial&, int8_t, int8_t, int8_t)':
c:\Users\Kio Store\Documents\Arduino\libraries\IBusBM\src\IBusBM.cpp:116:27: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  116 |         timer = timerBegin(timerid, F_CPU / 1000000L, true); // defaults to timer_id = 0; divider=80 (1 ms); countUp = true;
      |                 ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Kio Store\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal.h:84,
                 from C:\Users\Kio Store\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/Arduino.h:36,
                 from c:\Users\Kio Store\Documents\Arduino\libraries\IBusBM\src\IBusBM.cpp:25:
C:\Users\Kio Store\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
c:\Users\Kio Store\Documents\Arduino\libraries\IBusBM\src\IBusBM.cpp:117:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  117 |         timerAttachInterrupt(timer, &onTimer, true); // edge = true
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Kio Store\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
c:\Users\Kio Store\Documents\Arduino\libraries\IBusBM\src\IBusBM.cpp:118:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  118 |         timerAlarmWrite(timer, 1000, true);  //1 ms
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
c:\Users\Kio Store\Documents\Arduino\libraries\IBusBM\src\IBusBM.cpp:119:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  119 |         timerAlarmEnable(timer);
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm

Using library IBusBM at version 1.1.4 in folder: C:\Users\Kio Store\Documents\Arduino\libraries\IBusBM 
exit status 1

Compilation error: exit status 1`

fanfanlatulipe26, 2024-07-30:
Which platform? 

Which source / example ? 

Did you try with a previous release for ESP32 ?

 It seems you use 3.0.1

Have a try with 2.0.14

fanfanlatulipe26, 2024-07-31:
I am speaking about the software release of the ESP32 environment for Arduino. Please, try to install a previous release such as 2.0.14

fanfanlatulipe26, 2024-07-31:
The problem is with the "new" release of Arduino ESP32 core 3.0.1 that you are using.

Please have a look at [https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html)

In the timer section it is clearly stated that some API have changed or even been removed. Don't ask me why !!

So I hink that the best for you is to stay with a relese 2 of the ESP32 core.

As it is the IBusBM library for ESP32 board  is not compatible with ESP core 3.x.y 

Or you can try to not use a timer  (IBUSBM\_NOTIMER  and a call to  loop())

Too bad that so much incompatibilities were introduced .

yousif20121, 2024-08-01:
Already tried the IBUSBM_NOTIMER  & loop() method but it also didn't work.

After checking [](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html) they certainly changed some APIs which led to this incompatibility.

The downgrade of ESP32 core worked fine. I installed 2.0.14 and it uploaded normally without errors.

Thank you so much for your help.

fanfanlatulipe26, 2024-08-01:
Fine.
But be carefull when using the timer option with ESP32. See my issues #40 and #39 
The IBUSBM_NOTIMER should work

closed 2024-08-01 by yousif20121