bmellink/IBusBM #54: Does it work with megaAVG Nano Every?

CubeBike09, 2025-02-11:
I am new to Arduino and the IDE and I don't have lots of experience with coding. 
So, I wonder if this works with my megaAVG Nano Every. I need it for an RC control project.

I have, in the   Arduino IDE 2.3.4 simply imported the Library IBusBM and used some code from the example. I hope that is the right thing to do.

When compiling the code I get the message that this could be incompatible with the hardware.
Then compilation stops with "Compilation error: no matching function for call to 'IBusBM::begin(UartClass&)'"

What can be the reason for this? In the error messages I also read "reference to 'HardwareSerial' is ambiguous". 
As there is just one RX input on the Nano Every, is there something I need to change?
Thanks in advance for any hint.

fanfanlatulipe26, 2025-02-12:
Have a look at 

[https://github.com/bmellink/IBusBM/issues/16](https://github.com/bmellink/IBusBM/issues/16)
and  
[https://forum.arduino.cc/t/problem-using-ibusbm-with-nano-every/1061552](https://forum.arduino.cc/t/problem-using-ibusbm-with-nano-every/1061552)

In the "Issues" page you can make a search on all the issues (even if closed ....)

Hope it helps !

CubeBike09, 2025-02-16:
Thanks a lot - very helpful! This works now. :-)

Now I struggle to make my transmitter/receiver sending i-bus signals.The Flysky user manual for my FS-ST8 (with FS-SR8 receiver, I-bus capable) assumes one would connect an Serial Bus Receiver FS-CEV04. And for confirming activation of i-bus on the transmitter one should press a button on the FS-CEV04. But how to do that with the Arduino connected to the FS-SR8? How can I confirm the i-bus activation? 

From the manual of the transmitter:
"If the selected channel is about to assign to C1 channel of the Serial Bus Receiver FS-CEV04, then press the button K1 corresponding to C1 channel of FS-CEV04 receiver by a long thin tool. After the setting is successful, the system will pop up a menu showing successful."

fanfanlatulipe26, 2025-02-17:
Sorry, I don't have this equipment.
But as far as I understand a FS-CEV04 module is only required if you have more than 8ch in your model setup.
( the module takes a group of 4channels in the flow and output them on its C1-C4 plugs. You choose the group with a Kx button)
On the transmitter use the configuration for your  model , menu "Receiver Settings With INT RF /  RX SET - BIND SETTINGS" (??), you should be able to enable for example standard PWM output on the CH1-CH8 pins, and i-bus output on the SERVO plug. SENSE plug seems to be only i-bus sensor input.
Good luck ;-)